mercenaruss/uzg-firmware #1: Help me compile

ahonxxx, 2023-08-21:
[image]

mercenaruss, 2023-08-21:
@ahonxxx create a new folder: webh , inside src folder.
after:

Go to the folder tools/webfilesbuilder;
Run: npm install
Run: npx gulp

Then you will be able to build.

closed 2023-08-21 by mercenaruss

ahonxxx, 2023-08-21:
> @ahonxxx create a new folder: webh , inside src folder. after:
> 
> Go to the folder tools/webfilesbuilder; Run: npm install Run: npx gulp
> 
> Then you will be able to build.

Do you need to modify package.json   There are no files under webh

mercenaruss, 2023-08-21:
> 

no,just follow instruction above.

xyzroe, 2023-08-29:
> Still the same. There's nothing in the webh folder [image]

That's because you didn't run npx gulp command