mercenaruss/uzg-firmware #6: problems starting

SUNsung, 2023-11-10:
I used your firmware (UZG-01_v0.1.5.full.bin) for my
ZigStar UZG
After that the device stopped working

I wrote the firmware via USB using esptool (esptool.py write_flash 0x0 '/home/ad-ssd/Downloads/UZG-01_v0.1.5.full.bin')

The console endlessly writes this:
```
17:08:38.215 -> 10
17:08:38.415 -> [resetSettings] Start
17:08:41.402 -> [resetSettings] Led blinking done
17:08:41.435 -> [resetSettings] Config del done
17:08:41.475 -> Start
17:08:41.475 -> [hard reset] Entering hard reset mode
17:08:41.475 -> 1
17:08:41.667 -> 2
17:08:41.866 -> 3
17:08:42.065 -> 4
17:08:42.264 -> 5
17:08:42.462 -> 6
17:08:42.661 -> 7
17:08:42.861 -> 8
17:08:43.060 -> 9
17:08:43.260 -> 10
17:08:43.459 -> [resetSettings] Start
17:08:46.479 -> [resetSettings] Led blinking done
17:08:46.479 -> [resetSettings] Config del done
17:08:46.517 -> Start

```

mercenaruss, 2023-11-10:
Hello,
please open a ticket using your order number. https://help.zig-star.com/help/298083514 

Best Regards

mercenaruss, 2023-11-10:
Your device is a Chinese copy of LiliZig project. This firmware is not working with that device. Closing the issue.

closed 2023-11-10 by mercenaruss