mercenaruss/uzg-firmware #7: Zigbee restart needed every week

xmlguy74, 2023-11-24:
I have a UGZ-01 v0.2. I am fining that I need to restart the zigbee radio about every week in order to resolved a broken ZHA connection in home assistant.

When the issue occurs, I discover that HA shows a failure in the ZHA integration and all zigbee devices are unavailable. I am able to resolve the issue by using the USG-01 device web portal and clicking the System Tools -> Modules control -> zigbee restart button. After this, HA is able to reload the ZHA successfully.

This leads to think it's an issue with the zigstar device and not HA.

mercenaruss, 2023-11-24:
Hi @xmlguy74 ,please open a ticket with me by sending an email to support@zig-star.com.
or https://help.zig-star.com/help/298083514 .
I will assist you with this.

mercenaruss, 2023-11-24:
Please dont forget to add logs of ZHA to understand more the problem in your case.

closed 2023-11-24 by mercenaruss

BWilky, 2023-12-18:
Having the exact same issue

nicolinuxfr, 2024-02-16:
I think I have the same issue, but using Zigbee2MQTT. Unless I click on the restart ZigBee button in the UZG-01 web interface, Z2M won't start and the ZigBee devices are all unavailable.

I'm not sure what logs are useful ? Should I open a new issue @mercenaruss ?

```
error 2024-02-16 14:53:07: Error while starting zigbee-herdsman
error 2024-02-16 14:53:07: Failed to start zigbee
error 2024-02-16 14:53:07: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-02-16 14:53:07: Exiting...
error 2024-02-16 14:53:07: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

xyzroe, 2024-02-16:
It looks like you have known issue with Zigbee chip CCC2652P7. At the moment you couldn't do anything with it. Just need some time.

xyzroe, 2024-02-16:
yep

hilsonp, 2024-03-22:
I'm getting the same problem (https://github.com/xyzroe/ZigStarGW-FW/issues/27).

@xyzroe Do you have any reference of the forementionned firmware bug so we can follow/comment/help on this one ?

Couldn't the ESP32 detect the problem and restart the ZigBee chip ? (I would not care if you don't enable this tricky behaviour by default and let us enable it with a manual setting somewhere)

If you do not agree that such a work-around should be put in place until a firmware solution comes out, could the device expose a REST endpoint (basicauth is enough) to let us restart the zigbee module from home-assistant ?

I might try the ESPHOME Zigbee route but if I remember well, you suggest to stick to the stock ESP32 Zigstar firmware.

Can you please advise ?
We just cannot have the full network goes down without any way to automate the recovery.

I think, this case should stay open until a solution exists for the product.
I know that this is not directly under your control (as this is a firmware bug) but since this is a dependency for the product to work correctly, having the bug open would make other user find the information more easily.

xyzroe, 2024-03-22:
Problem is not related to ESP32, why it should be opened?

xyzroe, 2024-03-22:
Koenkk/Z-Stack-firmware#496

hilsonp, 2024-03-22:
Thank you for the link to the firmware issue.

I understand it is not related to the ESP32.
It is nevertheless related to the UZG-01 product (which uses the firmware).
From a buyer/user perspective, I would find it useful to be able to remotely ask the ESP32 module to reset the Zigbee chip.

The ESP32 module can do it (and exposes this action in the web page).
It would be nice if the ESP32 could detect this situation and reset the module itself.
If you think this is not acceptable or feasible, it would be nice to be able to trigger the action with a REST call, or a MQTT message.

Don't you think so ?

xyzroe, 2024-03-22:
REST call and a MQTT message are implement already 

https://github.com/mercenaruss/uzg-firmware/blob/5ba012831c0c8579f7d293828997f3a6850e25a2/src/web.cpp#L160

https://github.com/mercenaruss/uzg-firmware/blob/5ba012831c0c8579f7d293828997f3a6850e25a2/src/mqtt.cpp#L199

xyzroe, 2024-03-22:
From a buyer perspective you could open support ticket in zig-star.com and somebody will help you. Or just ask your seller.

xyzroe, 2024-03-22:
I have no idea how could ESP32 detect CC2652 hangup. But project is open source, so everyone is welcome to contribute 🙂

hilsonp, 2024-03-22:
> From a buyer perspective you could open support ticket in zig-star.com and somebody will help you. Or just ask your seller.

I saw I could open the ticket over there but thought it was better for the community to take the time to document it here.
But, thanks for the information. I'll follow your suggested process if I get another issue.

For the time being, I'll wait for the firmware fix (or help debug when a beta firmware is available for the P7).