mercenaruss/uzg-firmware #8: ERROR: cannot access local variable 'chip' where it is not associated with a value

olylak, 2023-12-16:
Hello,

I'm following instruction https://uzg.zig-star.com/flashing-and-updating/#multi-tool-ethernet to flash UZG-01 based on the CC2652P chip with the following firmware: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip

I'm getting the following error:
`local variable 'chip' referenced before assignment`

[image]
[image]

Also tried using cc2538-bsl with the same result:

`py ./cc2538-bsl.py -evw --ieee-address 00:12:4b:00:14:b5:56:da -p COM8 ./CC1352P2_CC2652P_launchpad_coordinator_20230507.hex`

Could you please advise what is the problem here?

Thanks.

xyzroe, 2023-12-16:
Could you try this one https://github.com/xyzroe/ZigStarGW-MT/releases/tag/v0.4.0 ?

But check twice which chip do you have BEFORE flashing. Don't look at ZigStar web UI CC2652 version and flash size - these field are just constant text. There is not any chip identifying function in current version of firmware.

olylak, 2023-12-17:
Thanks @xyzroe!  
Just opened a shekk and it's RF-Star RF-BM-2652P41 - CC2652P7. Will try 0.4.0 and let you know.

olylak, 2023-12-17:
Same result,

[image]

olylak, 2023-12-17:
Ok, looks like this pull request explains everything
https://github.com/JelmerT/cc2538-bsl/pull/149

xyzroe, 2023-12-17:
@olylak I flashed CC2652P7 using v 0.4.0 of MultiTool. Have you succeed with that?

olylak, 2023-12-17:
> @olylak I flashed CC2652P7 using v 0.4.0 of MultiTool. Have you succeed with that?

Nope, as I posted earlier - same result with version 0.4.0. Then I found pull request https://github.com/JelmerT/cc2538-bsl/pull/149 with changes to the cc2538-bsl script and re-flahed device successfully.

xyzroe, 2023-12-17:
There was error is windows version v0.4.0. it was fixes in v0.4.1 - community repost that app works as expected. 
So if you need you can try)

closed 2023-12-17 by xyzroe