mercenaruss/uzg-firmware #9: Status or health endpoint to programatically read status values

skrue, 2023-12-22:
It would be good to have an endpoint like `/health` or `/status` that allows for programmatic reading of the status values that are currently displayed in human-readable format on the `Status` HTML page. This way, external monitoring tools could easily be used to query UZG's status and configure alarms.

As a bonus, different formats could be provided, think `/status?format=json`, `/status?format=prometheus` or `/status?format=plaintext`, defaulting to `format=json`.

Another bonus would be having two separate endpoints, e.g. `/health` just returning `HTTP 200` and `OK` if the device is up and running without errors, and `/status` returning all the values that can currently be seen on the front page.

The HTTP status code should in any case reflect the actual state of the device.

Thank you for considering this feature request!

ffries, 2023-12-24:
Latest beta firmware allows to monitor the UZG from Home Assistant. If you are interested, please join the Discord channel and ask for the beta. It is running in my house without a problem for now 2 months.

skrue, 2024-01-04:
I have now looked into the latest beta (where the `/zha-z2m` page was empty, so I rolled back) and the response headers. For the latter, I was able to retrieve the JSON via curl: `curl -s -o nul -IL -w %header(respValuesArr) -u "user:pass" "http://IP-ADDRESS/api?action=0&page=0"` (Windows, Linux syntax will differ slightly).
Since my monitoring software (currently using Uptime Kuma) is not able to do custom `curl` commands, I ended up pulling the 0.1.5 sources from this repo and modifying `web.cpp` to include a status endpoint myself. To achieve this, I simply pulled out all the data acquisition part from the `void handleRoot()` function into a separate `DynamicJsonDocument getRootData()` function, that could then be reused in a new `void handleStatus()` function (that re-uses the authentication part from the `void handleApi().` function so no status information can be retrieved without authentication (if configured). It works perfectly fine for me.
I'd be happy to share this by opening a PR, but I wasn't sure which repo I should use.

BTW, how is the monitoring in HA supposed to work in the beta?

xyzroe, 2024-01-05:
You can make PR to my fork and I will PR to main repository when any updates will be ready. 

Monitoring in HA works using MQTT.

skrue, 2024-01-09:
Here's my PR: https://github.com/xyzroe/uzg-firmware/pull/2

xyzroe, 2024-01-09:
Thanks a lot. I will release new version in the nearest future and it will include you enhancements.

closed 2024-01-09 by xyzroe