mercenaruss/uzg-firmware #14: Config MQTT (auto discovery)

Razzer90, 2024-01-13:
hi, thanks for the great update. i'm just a bit at a loss as to how i can display the mqtt as sensors in home assistant? or restart the gateway via mqtt? i can see the individual mqtt states from the uzg in the mqtt explorer. but unfortunately there is no auto discovery in home assistant. thanks again!

xyzroe, 2024-01-13:
Did you enable toggle that called "Discovery" ?

Razzer90, 2024-01-13:
yes the switch is on, i also see everything in mqtt explorer, but mqtt has no entries from the uzg

xyzroe, 2024-01-13:
Just checked myself
[image]
[image]
[image]

Razzer90, 2024-01-14:
thanks for your help! i tried everything, home assistant reboot, uzg reflashed etc. the solution for me, i renamed the mqtt topic and it came up! perfect!

xyzroe, 2024-01-14:
What name topic had before rename and after?

Razzer90, 2024-01-14:
it was previously called Zigstar UZG-01, which did not work.
now simply UZG-01 and it worked immediately

xyzroe, 2024-01-14:
So it was a space in the topic name?

Razzer90, 2024-01-14:
yes, there was a space in between

xyzroe, 2024-01-14:
Add check input mqtt topic

dilighiani, 2024-02-01:
Can I run 2 instances of HomeAssistant (on 2 separate machines), one connects via ZHA and the other connects to the MQTT broker available on the UZG?

I tried first with ZHA, but then I am not able to make it to connect to the MQTT broker (enabled on the UZG).

dilighiani, 2024-02-01:
> connecting to the ZHA brings up Zigbee functionality but MQTT brings only internal states of your ZigStar. so the main purpose is different. what do you try to make?

Yes, now I re-read your words and I understand that the MQTT page brings ONLY the internal states of the device.

closed 2024-05-19 by mercenaruss