mercenaruss/uzg-firmware #15: Report the UZG-01 version or  CC2652P hw version accurately to help with zigbee firmware choice.

scubadam, 2024-01-25:
Given the stark warnings on picking the correct  CC2652P chip [here](https://uzg.zig-star.com/flashing-and-updating/#multi-tool-ethernet)
eg. CC2652P7 vs CC2652P2 

Would it not be possible to report this somehow to the user rather than hardcoding CC2652PXX
https://github.com/mercenaruss/uzg-firmware/blob/main/src/websrc/html/PAGE_ROOT.html#L87

I bought from elecrow and see no way of identifying what version I have.

Thanks this might make it easier for people to update their firmware thought there isn't actually a new one (currently 20230507 is the latest for either)

scubadam, 2024-01-25:
I saw that there is a rough rule that if there is a label on the bottom of the unit then its P7? (read on discord) perhaps that can be added to the firmware update docs to help users identify?

xyzroe, 2024-01-25:
I think it isn't related to firmware. 
Better place to put this information is official website with documentation. @mercenaruss What do you think?

joe2xyz, 2024-02-25:
Hi folks. Thank you for the awesome work!

Is it possible to get this info (CC2652P7 or CC2652P2) via CLI? Maybe https://github.com/JelmerT/cc2538-bsl can do that?

dbensmith, 2024-03-13:
Just adding my perspective: I went through the same process this week with a new UZG-01 from Elecrow. It came with ESP32 firmware v0.20 but the web console reports it like this: `0.2.0 (prod-solo)`. This sent me down a rabbit hole thinking the unit had the `ESP32-SOLO-1` chip and therefore was an early rev that needed ZigBee firmware for `CC2652P2`. When I flashed that version, however, ZigBee failed entirely.

Next I looked at ESP32 hardware revision. Since mine reported `ESP32-D0WDQ6`, I searched for information about this chip and found [this ESP32 reference](https://docs.espressif.com/projects/esp-idf/en/v3.3.5/hw-reference/modules-and-boards.html) that suggested my unit's ESP32 chip was not part of `ESP32-SOLO-1`. So maybe I had a new unit?

I tried flashing the firmware for `CC2652P7` and suddenly it started working and the TI chip's firmware version was then reported in the web console.

So I guess I have one of the newer revisions, but it took me a couple of hours to figure this out when it probably shouldn't need to. The main thing about prioritizing this is that this learning curve is only going to happen once per user, but it is a barrier to adoption for someone like me who screws up, but perhaps lack the time to figure it out later.

Are there any circumstances where UZG-01 units went out with ESP32/TI combinations _other_ than `ESP32-SOLO-1`/`CC2652P2` and `ESP32-D0WDQ6`/`CC2652P7`? If not, then can the web console assume the TI model based on the ESP32 model?

xyzroe, 2024-03-13:
@dbensmith please check v0.2.1

dbensmith, 2024-03-14:
> @dbensmith please check v0.2.1

[image]

Looks great, thank you for this update!

habitats-tech, 2024-03-29:
0.2.6 reports full info:

[image]

closed 2024-03-29 by mercenaruss