mercenaruss/uzg-firmware #21: Full flash is not possible due missing BOOT_APP0.BIN file

joe2xyz, 2024-02-24:
Hi @mercenaruss ,

I've just received ZigStar UZG-01, great product. Thank you!

I wanted to flash everything like this:
```
python3 ../tools/esptool.py --port /dev/ttyUSB0 write_flash --verify -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 UZG-01*.full.bin
```

But that's not possible since `boot_app0.bin` file is missing in `bin/` directory.

Could you please add it?

Thank you.

xyzroe, 2024-02-24:
You can use https://uzg.zig-star.com/webinstall/

xyzroe, 2024-02-24:
Also in our case you don't need boot_app0.bin

xyzroe, 2024-02-24:
Also keep in mind that UZG-01*.full.bin already includes bootloader_dio_40m.bin and partitions.bin

joe2xyz, 2024-02-25:
Thank you. I understand it now. I've just upgraded it via
```
python3 ../tools/esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --verify -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 UZG-01_v*.full.bin
```

joe2xyz, 2024-02-25:
Thank you. I understand now that UZG-01*.full.bin already includes everything.

closed 2024-02-25 by joe2xyz