mercenaruss/uzg-firmware #22: Zigbee not available: SRSP - AF - dataRequest after 6000ms

OleksandrChekmez, 2024-02-28:
Periodically Zigbee module becomes unavailable. 
Homeassistant Z2M shows the following errors in the logs: 
Failed to ping 'Boiler_switch' (attempt 1/1, Read 0xa4c138be83e96410/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))

The only solution is to restart Zigbee via System and Tools ->Restart Zigbee. 
It happens about once a week. I want to find the reason but don`t know how to find any logs that can bring some light. 

I checked Debug console and see the following:
`[1107887728] -> fe 37 24 01 1a 0c 01 01 00 ef d9 00 1e 2d 11 90 00 2e 00 65 00 00 24 08 00 26 0b 1e 26 12 00 22 17 00 1e 06 00 22 0c 00 22 0e 1e 22 17 00 22 06 00 22 0c 1e 22 0f 00 22 17 00 22 ca
[1107893734] -> fe 37 24 01 1a 0c 01 01 00 ef da 00 1e 2d 11 91 00 2f 00 65 00 00 24 08 00 26 0b 1e 26 12 00 22 17 00 1e 06 00 22 0c 00 22 0e 1e 22 17 00 22 06 00 22 0c 1e 22 0f 00 22 17 00 22 c9
[1108194675] -> fe 0f 24 01 97 8c 01 01 00 00 db 00 1e 05 10 93 00 00 00 72
[1108227324] | [ZBCHK] Wrong answer
[1108227324] | [ZBVER] Unknown`

Last 2 records appeared because I clicked Check Zigbee Connection. 

Device information:
Zig-Star model: | UZG-01
ESP32 fw ver: | 0.2.0 (prod-solo)
ESP32 hw version: | ESP32-D0WDQ6
ESP32 temperature: | 39.78 °C
ESP32 frequency: | 2 cores @ 160 MHz
ESP32 flash size: | 4 Mb, external
ESP32 heap (free / total): | 257 / 326 KiB
CC2652P hw version: | CC2652PXX

nicolinuxfr, 2024-02-29:
It might be the same issue as https://github.com/mercenaruss/uzg-firmware/issues/7

(maybe we should reopen this issue BTW ? Since it's pending a firmware update that's not available yet)

OleksandrChekmez, 2024-02-29:
yes, looks like the same issue.

closed 2024-03-21 by xyzroe