mercenaruss/uzg-firmware #24: UZG-01 POE v0.5 not connecting to ZHA

mpetcuRO, 2024-03-06:
Hi, followed all documentation but for some reason is not connecting to ZHA. I have another network Xiaomi coordinator that works correctly so I do not think is Home Assistant is to blame. Any ideas? did I get a bad one?

[Video](https://youtu.be/pqTPfpL7AaE)

mpetcuRO, 2024-03-07:
ok, tried with different network and a fresh VM of Home Assistant and works, so i guess is my installation of HA to blame. Any suggestions on what to try next? I really do not want to start over with HA I have 3 years of system configuration and devices. i do not see any error messages in HA logs so do not know where to start.

xyzroe, 2024-03-07:
HI! 
It's not a problem with UZG-01 because you see lots of messages in both direction in web console. 
Something goes wrong with your HA. I think you should look for some info about ZHA errors.

mpetcuRO, 2024-03-08:
ok, figure it out after countless try an error I narrowed it down to https redirection.
I host HA in docker under Synology NAS and have configured redirection http to https  and DNS. Once I accessed it using local IP without the redirection using 8123 port the UZG-01 installed with no error.

closed 2024-03-21 by xyzroe