mercenaruss/uzg-firmware #25: Cannot upgrade firmware

saintfrater, 2024-03-16:
Hello,

The web interface notify  about new ESP32 update available (v0.2.3).

I've tryed to upgrade via **System and Tools** -> **Firmware Upgrade** -> **Install later from github** 

the text change to *Github download starting...* but nothing change... 

I've also tryed via the manual procedure *"choose file..."* and click to *"ESP32 file update"* without other success.

here is the hardware/software information:

Zig-Star model:  UZG-01
ESP32 fw ver: 0.2.0 (prod-solo)
ESP32 hw version: ESP32-D0WDQ6
CC2652P hw version: CC2652PXX
CC2652P fw rev: 20230507

xyzroe, 2024-03-17:
Could you open Developer tools in your browser and then open Console. After that try to update firmware and note any messages in console. And attach them to your message.

saintfrater, 2024-03-17:
console log added to first comment.

xyzroe, 2024-03-18:
I don't see any errors. Possible causes: error in NTP time set from pool.ntp.org, inaccessible GitHub from ESP32.
I just checked update from 0.2.0 to 0.2.3 from GitHub all works perfect. [Video](https://vimeo.com/924653411?share=copy)

Something with your network. 
Also you could try to update manually using https://uzg.zig-star.com/webinstall/ and type C cable.

saintfrater, 2024-03-19:
indeed, I didn't see any request on my DNS servers....
Since I'm in "static IP", I did not find any way to setup the DNS servers, I think that the setting is missing on static IP setup, and so the UZG is unable to resolve any domains...

xyzroe, 2024-03-19:
Another possible problem is your timezone is very differs from mine, that could make problems with https. 
In v0.2.4 I added time zone selection.

saintfrater, 2024-03-21:
I confirm the upgrade works perfectly in DHCP mode... I'll open new "issue/request" for adding DNS setting when using "static IP" setting on ethernet.

closed 2024-03-21 by saintfrater