mercenaruss/uzg-firmware #27: Thread support for CC2652P7

q20, 2024-03-19:
Hey there

Any chance you could consider adding Thread support to your firmware?
The TI CC2652P7 supports:
` Bluetooth 5.2 Low Energy, Matter, Thread, Zigbee 3.0 `

It would be great to have another protocol enabled in Home Assistant by the UZG-01.  👍

xyzroe, 2024-03-19:
Hi.
It can be realized by changing CC2652 firmware, not ESP32. UZG-01 supports any protocol that is supported by CC2652.

q20, 2024-03-19:
> Hi. It can be realized by changing CC2652 firmware, not ESP32. UZG-01 supports any protocol that is supported by CC2652.

Is the functionality exposed by your firmware, though? Can Home Assistant use the Thread protocol with the device set up as per the instructions uzg.zig-star.com?

xyzroe, 2024-03-19:
I'm not very familiar with the Thread. But ESP32 receives bytes from cc2652 and transfers them to the socket connection, the same thing works in the other direction.  and it doesn’t matter which protocol runs on these bytes.

q20, 2024-03-19:
I think I understand: the firmware exposes everything from the TI chip. I guess I should check in at the maintainer of the Koenkk firmware...

xyzroe, 2024-03-19:
yeah you're absolutely right

closed 2024-03-19 by q20