mercenaruss/uzg-firmware #30: Add ability to reset network configuration using button

quaint-engineer, 2024-03-21:
It would a nice feature to reset all the network configuration using the rear button on the device.

I ran into an issue where I could not access the WebUI because I forgot to enter my subnet mask for a static IP. I attempted a factory reset but that didn't seem to reset the network section either.

I ended up attaching the gateway directly to my laptop via ethernet and assigning it an IP under the same subnet mask as the default value.

xyzroe, 2024-03-21:
But it should been already realized. 
Did you try long press (>10s) and didn't get succeed?

mercenaruss, 2024-03-23:
@xyzroe need to add by default 255.255.255.0 mask,many people forget to set this with static ip config,they assume is already there - because of grey color mask.

xyzroe, 2024-03-23:
@mercenaruss ok, but your comment is more related to #32

mercenaruss, 2024-05-19:
Assume this closed.

closed 2024-05-19 by mercenaruss