mercenaruss/uzg-firmware #31: Compiler prod-solo or prod

copilador, 2024-03-21:
Hello,
I want to compile the code but I'm not sure if I should use:

> [platformio]
> default_envs = prod-solo
> or
> [platformio]
> default_envs = prod

if I try to use default_envs = prod, should I add lib_deps = ESP Async WebServer, is that correct?"

xyzroe, 2024-03-21:
Based on chip you are using. But solo version on any esp32. 

No you shouldn't.

copilador, 2024-03-22:
The ESP32 on my board is an ESP32-WROOM. I believe the main difference between it and the ESP32-SOLO-1 is that the SOLO-1 has a single core, based on its datasheet (https://www.espressif.com/sites/default/files/documentation/esp32-solo-1_datasheet_en.pdf) (page4).
If I try to compile with

> [platformio]
> default_envs = prod

I need to include the option lib_deps = ESP Async WebServer. I'll do some more research to confirm. Thanks for your help!

closed 2024-03-22 by copilador

xyzroe, 2024-03-22:
you can use either solo or original firmware on ESP32 WROOM.