mercenaruss/uzg-firmware #32: Static IP configuration enhancement request

saintfrater, 2024-03-21:
can you add, ethernet configuration page, the abilities to specify other DNS Servers ?

olgru, 2024-03-22:
I'm still trying to find out which DNS is used by default. I'm not happy not being able to find basic information about DNS and NTP. These values should be configurable. I block all outgoing / incoming DNS which is not passing my own DNS setup. Same for NTP where I rely on an internal time source. This is a general risk mitigation recommendation.

xyzroe, 2024-03-22:
https://github.com/mercenaruss/uzg-firmware/blob/5ba012831c0c8579f7d293828997f3a6850e25a2/src/etc.cpp#L294

xyzroe, 2024-03-23:
https://github.com/mercenaruss/uzg-firmware/issues/30#issuecomment-2016455871

olgru, 2024-03-23:
> https://github.com/mercenaruss/uzg-firmware/blob/5ba012831c0c8579f7d293828997f3a6850e25a2/src/etc.cpp#L294

Thanks for that link - OK, we have pool.ntp.org and time.google.com. I have no problem with pool.ntp.org, but anything towards google.com is filtered on FW. I prefer to set this to a local time server - you know, if the local time is wrong everything local will be wrong the same way - which makes them all (relative to each other) working correctly (eg. key expire times etc). Using different time sources within a local network in parallel could make systems inaccessible if one time source is wrong.

olgru, 2024-03-23:
> Thanks for that link - OK, we have pool.ntp.org and time.google.com. I have no problem with pool.ntp.org, but anything towards google.com is filtered on FW. I prefer to set this to a local time [...]

> > I'm still trying to find out which DNS is used by default. I'm not happy not being able to find basic information about DNS and NTP. These values should be configurable. I block all outgoing / incoming DNS which is not passing my own DNS setup. Same for NTP where I rely on an internal time source. This is a general risk mitigation recommendation.
> 
> You're right, being able to change NTP is a very nice to have feature... I'll added as a sub request to this request in the main topic.
> 
> When you're using DHCP, the device will obey to your DHCP's DNS settings. When you're using Static IP, the DNS setting isn't set (left empty),

Sorry I'm not sure on your last sense. On static IP, there must be a default DNS as update checks are working - or it's really coded with IP addresses... Or we have a beautiful etc/hosts file... ;-) On my systems the DNS is a pihole cluster which I've allowed to be bypassed for this test. But usually within my networks there is no other DNS possible than mine...

saintfrater, 2024-03-23:
> Sorry I'm not sure on your last sense. On static IP, there must be a default DNS as update checks are working - or it's really coded with IP addresses... Or we have a beautiful etc/hosts file... ;-) [...]

Same for me, but I didn't override, I've just switch from static to DHCP and my DHCP server set the DNS to my DNS Servers.
I was saying that I do suspect that, on static IP, there is *no* DNS configuration set so update are failling, I haven't found, within the etc.cpp code, any reference to DNS setting when static configuration is enabled.

closed 2024-05-19 by mercenaruss