mercenaruss/uzg-firmware #35: better support for leds on v1 (CC2652P2)

532910, 2024-04-04:
BLUE / RED / YELLOW buttons are shown for v1, though they don't change anything

[image]

xyzroe, 2024-04-04:
FW version?

532910, 2024-04-05:
the latest one, of course:
ESP32 fw ver: 	0.2.6 (prod-solo)
ESP32 hw version: 	ESP32-D0WDQ6
CC2652P hw version: 	CC2652P2
CC2652P fw rev: 	20230507

xyzroe, 2024-04-11:
Please provide original express.com link where exactly attach your photos

532910, 2024-04-11:
I will attach photos.

xyzroe, 2024-04-12:
You are using HamGeek that is not original UZG. Maybe there are some differences from UZG schematics.

532910, 2024-05-06:
[image]
[image]

mercenaruss, 2024-05-19:
This is not a ZigStar Device. Hamgeek maybe

closed 2024-05-19 by mercenaruss

532910, 2024-05-20:
It's hamgeek. No plan to support it?

mercenaruss, 2024-05-21:
No,sorry.

xyzroe, 2024-05-21:
there is only a Zigbee LED in your photos. So there is nothing to work from buttons in the UI.

532910, 2024-05-21:
I meant, is it possible to detect board modification from the software and hide buttons in the UI?