mercenaruss/uzg-firmware #36: V0.2.6 on CC2652P7

NightHawkATL, 2024-04-10:
I have finally updated to the "latest" version and I am still seeing daily disconnects with my Home Assistant setup.

I have tried:
- Rebooting the ESP32 module
- Rebooting the Zigbee module
- Cycling the port on the switch it is connected to
- Reloading the integration
- Rebooting Home Assistant

Is there anything that can be done to make this more stable or should I just move to a competitor device? This has only happened in the last month or so and was rock solid before that.

xpenno255, 2024-04-10:
I'm also having issues that sounds similar to this. For me it happens every 2 to 3 days.

I'm using ZHA in Home Assistant OS 2024.04.2 I was previously running Skyconnect via USB but I migrated to UZG-01 in ethernet mode and updated directly to v0.2.6.

At the time the following error appears in home assistant.

` Logger: zigpy.application
Source: /usr/local/lib/python3.12/site-packages/zigpy/application.py:675
First occurred: 20:24:14 (1 occurrences)
Last logged: 20:24:14

Watchdog failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/api.py", line 1059, in request
    response = await response_future
               ^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 673, in _watchdog_loop
    await self.watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 655, in watchdog_feed
    await self._watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/zigbee/application.py", line 628, in _watchdog_feed
    await self._znp.request(c.SYS.Ping.Req())
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/api.py", line 1055, in request
    async with async_timeout.timeout(
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

`

NightHawkATL, 2024-04-10:
@xpenno255 i just recently added the MQTT server connection to add functions to monitor it in HA. I have an automation that watches the state of ZHA and if it goes to unavailable or unresponsive, it sends a message and then triggered the zigbee reboot of the UZG-01. Once it comes back up, it sends another message letting me know it may be working again.

Trying this for a minute to see if it helps.

tavalin, 2024-04-10:
Can you share your automation or a link to it please? I'm having similar issue

NightHawkATL, 2024-04-10:
@tavalin [image]

NightHawkATL, 2024-04-10:
> Can you share your automation or a link to it please? I'm having similar issue 

```
alias: "Notify: Zigbee Control Down"
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.texas_instruments_coordinator
    from: null
    to: unavailable
condition:
  - type: is_not_connected
    condition: device
    device_id: aeb436d05cd0bbf44cac48c2043ce89c
    entity_id: de5d3f881a20242a52e9f54da7b74635
    domain: binary_sensor
action:
  - service: notify.mobile_iphone_13
    metadata: {}
    data:
      message: >-
        The Zigbee coordinator has gone unresponsive. The zigbee radio has been
        restarted and should become responsive shortly.
      title: Notice!
  - device_id: aeb436d05cd0bbf44cac48c2043ce89c
    domain: button
    entity_id: 3affbf71bed59a894f70b0c1c110753b
    type: press
mode: single

```

NightHawkATL, 2024-04-10:
[image]
[image]
[image]
[image]

xyzroe, 2024-04-11:
https://github.com/mercenaruss/uzg-firmware/issues/7

xyzroe, 2024-04-11:
It's some problem with CC2652 firmware. You could find more information https://github.com/Koenkk/Z-Stack-firmware/discussions/496

NightHawkATL, 2024-04-11:
> It's some problem with CC2652 firmware. You could find more information https://github.com/Koenkk/Z-Stack-firmware/discussions/496

There's no clear fix on that site. It is just a lot of trial and error so far. I have my automation in place until they can come up with a fix.

xpenno255, 2024-04-12:
> ![Screenshot 2024-04-10 7 01 54 PM](https://private-user-images.githubusercontent.com/8395658/321426490-8f80f6c4-bb02-4061-ba96-59063cf3f84d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDg4MzAsIm5iZiI6MTcxMjk0ODUzMCwicGF0aCI6Ii84Mzk1NjU4LzMyMTQyNjQ5MC04ZjgwZjZjNC1iYjAyLTQwNjEtYmE5Ni01OTA2M2NmM2Y4NGQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTJUMTkwMjEwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTBhOWJiZGUxZDQ0OTViNDI1NjE4ODhlZDEwNDc0YTliY2VkNTY5MzU1YzczOWUzNmYxOTFhNDhlMTAxODNkMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.YXIifGzLhrX-NeOqGapdcYgJMWwdVp_vNm98epJ96LQ) ![Screenshot 2024-04-10 7 00 05 PM](https://private-user-images.githubusercontent.com/8395658/321426669-9e93235b-dfd3-45f2-aa97-fa8915701e82.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDg4MzAsIm5iZiI6MTcxMjk0ODUzMCwicGF0aCI6Ii84Mzk1NjU4LzMyMTQyNjY2OS05ZTkzMjM1Yi1kZmQzLTQ1ZjItYWE5Ny1mYTg5MTU3MDFlODIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTJUMTkwMjEwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGQwNWMxNGNiNGNkNjM4NTk0MTQ1YzNlZTQyYTc2MzE2YjNjZWMwZTg2ODhiMDQyY2MwNTNkOWVmYmMyZmNjZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3h6VgPKR9E1RUyOFXWT3XTeD1S4uwxyukGqFrcKHBGI) ![Screenshot 2024-04-10 7 01 13 PM](https://private-user-images.githubusercontent.com/8395658/321426703-178edc2f-d6a1-4a28-b584-6896b8b59425.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDg4MzAsIm5iZiI6MTcxMjk0ODUzMCwicGF0aCI6Ii84Mzk1NjU4LzMyMTQyNjcwMy0xNzhlZGMyZi1kNmExLTRhMjgtYjU4NC02ODk2YjhiNTk0MjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTJUMTkwMjEwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTU2ZGU3YWM1ODA4MTBmNDFmM2I0NjgzMzhhYzdjMTMxNjUwMDNiYzg2MWE2ZTA3MTNjMzRhM2ZlOGM3M2E0MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tzpsc0I-SGhY0TP28o9R710f9BuAtg0Zq8Xqbvm9-18)

This looks great, I'm currently using ethernet mode to connect my UZG-01 so I'm not sure if it appears as entity in home assistant, I can't seem to find it at least....

xyzroe, 2024-04-12:
we don't see you images 😞

NightHawkATL, 2024-04-12:
I downloaded the files linked at the top labeled 20240315 and extracted them and installed it on my UZG-01 and I show "20230507" but it is stable as of right now. I have not gotten any more disconnect notices from my automation to monitor it since the update and connections do seem to be faster and more reliable.

NightHawkATL, 2024-04-12:
> This looks great, I'm currently using ethernet mode to connect my UZG-01 so I'm not sure if it appears as entity in home assistant, I can't seem to find it at least....

You will need to enable the MQTT connection in the latest ESP32 update to get access to the reset commands. But by default, you can use the ZHA integration to track the status as shown in my screenshots. @xyzroe I think he was linking to my screenshots.

TronTronix, 2024-04-28:
Ich versuche bei mir ein updat für CC2652P7 einzuspielen (ZigStarGW-MT) bleibt aber immer stehen mit ("local variable cip referenced before assignment")
[image]

Mit (SmartRF FP2 - USB) hat es auf jeden Fall funktioniert
[image]

Aber jetzt habe ich irgendwie wohl eine Dateileiche ("w.hex") im System wie kann ich die loswerden, löschen ??
[image]

xyzroe, 2024-05-01:
> Ich versuche bei mir ein updat für CC2652P7 einzuspielen (ZigStarGW-MT) bleibt aber immer stehen mit ("local variable cip referenced before assignment") ![1](https://private-user-images.githubusercontent.com/122129450/326291303-d95ae779-3477-452f-bfe0-af6d54bb54c5.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1NTAzOTYsIm5iZiI6MTcxNDU1MDA5NiwicGF0aCI6Ii8xMjIxMjk0NTAvMzI2MjkxMzAzLWQ5NWFlNzc5LTM0NzctNDUyZi1iZmUwLWFmNmQ1NGJiNTRjNS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMVQwNzU0NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iY2E2NmE3ZDUyMTI0ZWY4NWFjY2E0NWRkY2I4NzBkNjg4MWJmMzY0NWVjMWU4YmE4OTcxMmIxN2JjN2U2MzJlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.J2cKJG4p9oGq36MOAcDj751G8M440VNav-PPTSMmG6I)
> 
> Mit (SmartRF FP2 - USB) hat es auf jeden Fall funktioniert ![4](https://private-user-images.githubusercontent.com/122129450/326292024-0d7191d4-68f9-4eb2-abef-a805a739a4b9.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1NTAzOTYsIm5iZiI6MTcxNDU1MDA5NiwicGF0aCI6Ii8xMjIxMjk0NTAvMzI2MjkyMDI0LTBkNzE5MWQ0LTY4ZjktNGViMi1hYmVmLWE4MDVhNzM5YTRiOS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMVQwNzU0NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZTYyMDljYmU0MjZiMTQ4YzZkZmNmMjMxMzFhYjU2ZDQ5YTViZTYzNzU1ODhmYWM2NzRmYTE0OTI3MmRhZGNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.53jauYIehBmiSEWdNjlakwBOD7YZg4dCD74PoDrOZOI)
> 
> Aber jetzt habe ich irgendwie wohl eine Dateileiche ("fw.hex") im System wie kann ich die loswerden, löschen ?? ![2](https://private-user-images.githubusercontent.com/122129450/326291435-8f766a12-3d17-4e96-b0ea-ad563ba7e773.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1NTAzOTYsIm5iZiI6MTcxNDU1MDA5NiwicGF0aCI6Ii8xMjIxMjk0NTAvMzI2MjkxNDM1LThmNzY2YTEyLTNkMTctNGU5Ni1iMGVhLWFkNTYzYmE3ZTc3My5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMVQwNzU0NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNzM0MGU2NDUxMGJiOTNmNzBiZGM5MTc2ZTNiOGRhNzQ0MjVmZDNlMGNjNDc2ODc4MTgyOGZmMDEyMDVjNDFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Tl-hj_P2RHVJoN0ZUJNz2U2mAa6qXeAWKf_dVq-945Y)

1. You need to use latest version of Multitool. You were used old one. Need 0.4.0 at least.
3. This files appears after trying to install Zigbee update via web UI, which is still in development. There isn't way to remove this file except full erase flash and install clean firmware. 

P.S. Please be respectful of other community members, and use the language that everyone uses to communicate in this community.

GD-Dal, 2024-05-03:
Having the same issues.
And if I enable MQTT, it becomes downright unstable. Not able to connect at all
I use PoE to power it. But every time I initiate a reboot from the Zigstar device, the NIC just blinks and never comes online anymore, unti i do a shutdown / no shutdown on the switchport

Not particulary happy right now

mercenaruss, 2024-05-04:
@GD-Dal this will be solved,this is not a generic problem,cant find the pattern for this problem,never occurs on my test devices.
Please send a email to: support@zig-star.com with details about you data switch.

closed 2024-12-15 by xyzroe