mercenaruss/uzg-firmware #39: Socket client disconnects, requires ESP32 Restart

stormshaker, 2024-05-14:
My ZHA client disconnects from the UZG-01 about every two days and cannot reconnect when it retries. To fix it, I need to use the 'ESP32 Restart' button on the System Tools page of the UZG-01, and then reload ZHA.

[image]
The Raw Data log looks strange when the UZG-01 is in this state:
[image]

[image]

The next time this happens, what logs can I collect to help troubleshoot this issue?

itn3rd77, 2024-05-15:
Same problem on my side. Console full of hex 'ef' messages. A Zigbee Restart command ist required for client (Home Assistant) to reconnect.

BWilky, 2024-05-18:
Having this same issue as well. Disconnects constantly.

xyzroe, 2024-05-19:
@mercenaruss I think it caused by Zigbee chip. What do you think?

mercenaruss, 2024-05-19:
If you are using version with P7 chip inside,please update the firmware to latest one.
https://github.com/Koenkk/Z-Stack-firmware/discussions/496

closed 2024-12-15 by xyzroe