mercenaruss/uzg-firmware #42: Unable to update firmware

jamiebegin, 2024-05-26:
When I log into the web interface, I get a notification to upgrade to 1.0. However, I am unable to do so. Trying to upgrade directly from Github doesn't do anything. So I then downloaded the release and attempted to upload it via the web interface. However, when I select the file in the file dialog, it is input as "C:\fakepath\XZG_20240526.1.full.bin". So not surprisingly, it doesn't upload correctly there either.

xyzroe, 2024-05-26:
You could try to restart ESP32 and right after that click install from GitHub.
To use local file update you need to select https://github.com/mercenaruss/uzg-firmware/releases/download/v1.0.0/UZG-01.bin or https://github.com/xyzroe/XZG/releases/download/20240526.1/XZG_20240526.1.ota.bin
Not *.full.bin but *.ota.bin

closed 2024-05-26 by mercenaruss

jamiebegin, 2024-05-27:
Not sure why this was marked closed? The suggestion did not work. What ended up working was changing the timezone to UTC.

xyzroe, 2024-05-27:
What is your local time zone?

epiphanyplx, 2026-02-15:
Quick question, seems firmware update went bad (maybe, not really sure), lose 2-3 packets out of every 15 or so and can't access web page - seems to be reboot looping. Just blue light flashing.

Is there anything I can do that will allow me to keep the Zigbee config? 
Wasn't sure if the reflash you are mentioning will wipe that part or not. 
Home Assistant isn't able to communicate so can't backup from there. 

Have 56 devices so would love to not have to recreate.