mercenaruss/uzg-firmware #43: Fw Update 0.26 -> 1.0 = device completely bricked

jlestrohan, 2024-05-31:
I have tried to have the device update itself via the web interface (OTA), seemed stuck (no progression) so I decided to download the latest fw and upload it to the device... terribly wrong, the device is now totally dead except for a blue fixed led 

Now I can't update it at all. Tried everything, erase flash with esptool then reflash it, looks totally dead, i'm now the proud owner of a useless brick after 15 days of use. 

Any help would be very welcome to make that device work again (i'm on Linux, fedora 40)
The web installer fails everytime, flashing with esptool works but nada I can do to have this firmware work again.
Any special op to maybe rebuild the partition table ? flash to a certain address ?.... 

Many thanks in advance i'm very bummed :/

KillerKelvUK, 2024-06-01:
Wow glad I checked the issues before doing the same because of the same.  Hopefully the devs' will come back to you ASAP to help recover it.

xyzroe, 2024-06-01:
@jlestrohan, could you please provide photos of your device to help us identify the exact hardware revision? If possible, please open the case and take photos of the board. If you cannot open the case, photos of the device's bottom side will be helpful.

Additionally, please answer the following questions to assist us further:

1. How did you manually upload the update the first time? Which method did you use and what file was involved?
2. Could you provide a screenshot an error encountered during the web installer process?
3. Have you tried using a different PC to flash the device via the web installer?

jlestrohan, 2024-06-02:
Thank you @xyzroe for your reply :)

Device back sticker says "UZG-POE 0.5"
I bought it about one month ago from Tindie (order on May 12 2024 2,44 am)

1) From the web interface -> download last update from github -> stuck at 0%
Then I downloaded the UZG-01.bin from https://github.com/mercenaruss/uzg-firmware/releases/download/v1.0.0/UZG-01.bin and started a manual upload from the very same web interface
Rebooted and then no IP, no dhcp, no wifi, nada...
When I plugged it in USB and went to the fw update page of the official website I still could connect and get some infos from the device
2) When I try to update from the website (https://uzg.zig-star.com/webinstall/) after choosing ttYUSB0, I get the menu asking if I want to either Install UZG Fw or open a console, I then select Install FW and it hangs at preparing installation for like 10 mins and then "FGailed to Initialize (see sc)

I used esptools.py to try to erase then reflash witht eh full.bin firmware, flash said it completed with success but still unable to boot or do whatever... 

[image]
[image]
[image]

Tried from a Win10 PC from the web installer => same thing :/

xyzroe, 2024-06-02:
I have just checked. Take UZG-01 0.5 (Default version was 0.2.0) Just after boot up get notice about available 1.0.0 (XZG), go to Tools, click "install latest", and it installed successfully. Wait to boot up and get working 20240519. Then updated to 20240602 also using git install latest. Now I have 20240602, without any manual re-flashing. 

Next step I'll try to re flash using web flasher. Come back later

xyzroe, 2024-06-02:
I have successfully re-flashed using [XZG web flasher](https://xzg.xyzroe.cc/install/). Device works without any issues.

jlestrohan, 2024-06-03:
I achieved to reflash it using esptools, took me a while tho but now I reverted to 0.26 and will stay there for a while :)
Thanks all

closed 2024-06-03 by jlestrohan