mercenaruss/uzg-firmware #44: Local version v0.2.6 won't upgrade to GitHub version v1.0.0 via the update process in the UI

KillerKelvUK, 2024-06-01:
Click the "Install latest from Github" button, "Github download starting..." shows above the progress bar but then nothing happens, ever.  I was about to try uploading the ESP32 binary via the file update button but thankfully saw [this](https://github.com/mercenaruss/uzg-firmware/issues/43) issue.  Please can you advise on the upgrade process I need to follow.

FWIW...I'm using a wired network with a statically assigned IP, have enabled webserver authentication, set a custom hostname and changed the timezone.

I could try a reset before the update process but happy to provide some logging/testing before that if it will help.

Thanks

xyzroe, 2024-06-01:
@KillerKelvUK
1. Please try restarting your device and then attempt the "Install latest from GitHub" option again.
2. To perform a successful manual update, download the *.ota.bin file and select it in the file update dialog within the web UI.

KillerKelvUK, 2024-06-02:
Thanks, 1 didn't work but 2 did.

closed 2024-06-02 by KillerKelvUK