mercenaruss/uzg-firmware #47: Ethernet stopped working on both UZG-01

ezcGman, 2024-07-24:
Hey there,

I don't know what to do anymore... I ordered two UZG-01 from Elecrow which arrived few days ago. I hooked up the first one via USB and flashed it via the ESP Web Flasher to the latest 0.x version (web flasher only does 0.x). I then hooked it up to via a PoE supported Ethernet port and it all worked fine. It powered up, got an IP, all good.
I flashed the Zigbee Router firmware, got it hooked up to ZHA, started adding devices, all good so far.
However, the device stopped responding to http / via ethernet after some time, like it would go into wifi sleep... Zigbee was still working. Reconnecting the ethernet cable fixed it eventually. However, after some time it complete stopped even using Ethernet: The LEDs next to the Ethernet port don't light up, BUT it powers up via PoE! Also Zigbee seems to work still. I hooked it up to my Laptop again, flashed the latest 1.0.0 (UZG-01.bin) via ESP Flasher: Nothing, no change. I however see the WiFi AP it broadcasts. I can connect to it, but it refuses to even use Ethernet...
SOMETIMES, I alternatively get both LEDs lighting up completely, no flashing. Just full on.

So I got the 2nd one out, directly flashed 1.0.0 (this time XZG_20240519.full.bin, as I think it contains a few partitions?) via Webflasher, hooked it up via Ethernet: Works! Nice! Flashed the router firmware again, hooked it up to my coordinator, gave it a new IP (yes, I gave it a new IP before that and it surviced it), reconnected cable aaaaaand same issue: No lights at all on the Ethernet port... It survived several reboots, several reconnects, but all of the sudden stops working on Ethernet...

I took the first one back to the Laptop, erased flash via esptool, flashed XZG_20240519.full.bin vie ESP Flasher: Nothing, no Ethernet lights.

I even tried hooking it up via USB and then connect Ethernet: Same story. One time, when I got the case where instead both LEDs just light up, I disconnected the Ethernet cable and the LEDs were still on :D See picture below:
[image]

I really don't know what to do anymore. Are they broken? Any hints to get them back?

Thanks and greetings,

Andy!

ezcGman, 2024-07-24:
After posting, I hooked it up again via Ethernet aaaand it worked... Still old IP it has not assigned on DHCP server. Did a factory reset, still old IP... Flashed Zigbee firmware again, it auto rebooted, didn't come back. I went looking at device: both Ethernet LEDs off. Reconnected wire: works... Still old IP. It refuses to get the IP it has assigned by the DHCP... Is there a way to make it forget it? Maybe that is somebody cause an issue? Still, wouldn't explain why the Ethernet port LEDs don't light up at all...

ezcGman, 2024-07-25:
Aaaaand again: I disconnected it from Ethernet, re-connected it aaaand no LEDs light up, not reachable via LAN, BUT when connecting to it via Wi-Fi-AP, it shows the new IP I assigned yesterday (so somehow it was able to fetch it), but says LAN: disconnected (which it actually is...)

ezcGman, 2024-07-25:
If it helps, I hooked it up via USB to get the logs, these are the bootup logs after restarting it from the web interface (while being connect via Wi-Fi AP):

```
[13:22:22][35m1156 [0m[32m[printLogMsg] [0m[ZB] Chip: CC2652P7
[13:22:22][35m1157 [0m[32m[printLogMsg] [0m[ZB] IEEE: 00:12:4B:00:2E:13:2B:8D
[13:22:22][35m1157 [0m[32m[zbHwCheck] [0mmodeCfg ffc1b9fd
[13:22:22][35m1167 [0m[32m[zbHwCheck] [0mbslCfg c50ffec5
[13:22:22][35m1167 [0m[32m[printLogMsg] [0m[ZB] Flash size: 704 KB
[13:22:23][35m1734 [0m[32m[setup] [0mConfig:
[13:22:23]{
[13:22:23]  "network-config": {
[13:22:23]    "wifiEnbl": false,
[13:22:23]    "wifiSsid": "",
[13:22:23]    "wifiPass": "",
[13:22:23]    "wifiDhcp": true,
[13:22:23]    "wifiIp": "::",
[13:22:23]    "wifiMask": "255.255.255.0",
[13:22:23]    "wifiGate": "::",
[13:22:23]    "wifiDns1": "1.1.1.1",
[13:22:23]    "wifiDns2": "8.8.8.8",
[13:22:23]    "wifiPwr": 78,
[13:22:23]    "wifiMode": 1,
[13:22:23]    "ethEnbl": true,
[13:22:23]    "ethDhcp": true,
[13:22:23]    "ethIp": "::",
[13:22:23]    "ethMask": "255.255.255.0",
[13:22:23]    "ethGate": "::",
[13:22:23]    "ethDns1": "1.1.1.1",
[13:22:23]    "ethDns2": "8.8.8.8"
[13:22:23]  },
[13:22:23]  "vpn-config": {
[13:22:23]    "wgEnable": false,
[13:22:23]    "wgLocalIP": "::",
[13:22:23]    "wgLocalSubnet": "255.255.255.0",
[13:22:23]    "wgLocalPort": 33333,
[13:22:23]    "wgLocalGateway": "0.0.0.0",
[13:22:23]    "wgLocalPrivKey": "",
[13:22:23]    "wgEndAddr": "",
[13:22:23]    "wgEndPubKey": "",
[13:22:23]    "wgEndPort": 0,
[13:22:23]    "wgAllowedIP": "0.0.0.0",
[13:22:23]    "wgAllowedMask": "0.0.0.0",
[13:22:23]    "wgMakeDefault": true,
[13:22:23]    "wgPreSharedKey": "",
[13:22:23]    "hnEnable": false,
[13:22:23]    "hnJoinCode": "",
[13:22:23]    "hnHostName": "XZG-8467",
[13:22:23]    "hnDashUrl": "default"
[13:22:23]  },
[13:22:23]  "mqtt-config": {
[13:22:23]    "enable": false,
[13:22:23]    "server": "",
[13:22:23]    "port": 1883,
[13:22:23]    "user": "",
[13:22:23]    "pass": "",
[13:22:23]    "topic": "XZG-8467",
[13:22:23]    "updateInt": 60,
[13:22:23]    "discovery": true,
[13:22:23]    "reconnectInt": 15
[13:22:23]  },
[13:22:23]  "system-config": {
[13:22:23]    "disableWeb": false,
[13:22:23]    "webAuth": false,
[13:22:23]    "webUser": "",
[13:22:23]    "webPass": "",
[13:22:23]    "fwEnabled": false,
[13:22:23]    "fwIp": "0.0.0.0",
[13:22:23]    "serialSpeed": 115200,
[13:22:23]    "socketPort": 6638,
[13:22:23]    "tempOffset": 32,
[13:22:23]    "disableLedUSB": false,
[13:22:23]    "disableLedPwr": false,
[13:22:23]    "refreshLogs": 1,
[13:22:23]    "hostname": "XZG",
[13:22:23]    "timeZone": "Europe/Berlin",
[13:22:23]    "ntpServ1": "pool.ntp.org",
[13:22:23]    "ntpServ2": "time.google.com",
[13:22:23]    "nightMode": false,
[13:22:23]    "startHour": "23:00",
[13:22:23]    "endHour": "07:00",
[13:22:23]    "workMode": 0,
[13:22:23]    "zbRole": 0,
[13:22:23]    "zbFw": "?",
[13:22:23]    "updHour": "01:00",
[13:22:23]    "updDays": "*",
[13:22:23]    "autoIns": false
[13:22:23]  }
[13:22:23]}
[13:22:23][35m1890 [0m[32m[setup] [0mVARS:
[13:22:23]{
[13:22:23]  "system-vars": {
[13:22:23]    "hwBtnIs": true,
[13:22:23]    "hwLedUsbIs": true,
[13:22:23]    "hwLedPwrIs": true,
[13:22:23]    "hwZigbeeIs": true,
[13:22:23]    "connectedClients": 0,
[13:22:23]    "socketTime": 0,
[13:22:23]    "connectedEther": false,
[13:22:23]    "apStarted": false,
[13:22:23]    "wifiWebSetupInProgress": false,
[13:22:23]    "vpnWgInit": false,
[13:22:23]    "vpnWgConnect": false,
[13:22:23]    "vpnWgPeerIp": "::",
[13:22:23]    "vpnWgCheckTimeKey": 0,
[13:22:23]    "vpnHnInit": false,
[13:22:23]    "mqttConn": false,
[13:22:23]    "mqttReconnectTime": 0,
[13:22:23]    "mqttHeartbeatTime": 0,
[13:22:23]    "disableLeds": false,
[13:22:23]    "zbFlashing": false,
[13:22:23]    "deviceId": "XZG-8467",
[13:22:23]    "espUpdAvail": false,
[13:22:23]    "zbUpdAvail": false
[13:22:23]  }
[13:22:23]}
[13:22:23][35m1952 [0m[32m[setupCoordinatorMode] [0mNetwork
[13:22:27][35m6135 [0m[32m[mDNS_start] [0mmDNS responder started on XZG.local
[13:22:27][35m6148 [0m[32m[printNVSFreeSpace] [0mTotal: 630, Used: 132, Free: 498
[13:22:39][35m17808 [0m[32m[printLogMsg] [0m[ZB] FW: Unknown! Check serial speed!
[13:22:39][35m17809 [0m[32m[setup] [0m[ESP] FW: 20240707
[13:22:39][35m17809 [0m[32m[setup] [0mLoad cfg UZG-01
[13:22:40][35m18572 [0m[32m[setup] [0mdone
```

ezcGman, 2024-07-25:
Yeah, I give up...
- I tried three different cables from two different brand: No ethernet 98% of the time, but 100% always powering up via PoE
- I tried hooking it up to the switch directly in the basement (so far always to a PoE port of an wifi access point: same...
- I tried flashing the old firmware via https://uzg.zig-star.com/webinstall/: also no Ethernet

I don't know, something is not right, I'm pretty close to ask for a refund / returning those, sadly...

xyzroe, 2024-07-26:
If you connected USB while LAN POE is enabled, it can damage LAN chip.

ezcGman, 2024-07-26:
Yeah, I have been in a remote session with @mercenaruss and something is off / broken.

I never hooked up the UZG to a PoE port when powering it also via USB. I chose the PoE port only when not hooking it to USB.

xyzroe, 2024-07-27:
Ok. @mercenaruss will help you) 
I close issue as it is not related to firmware.

closed 2024-07-27 by xyzroe