mercenaruss/uzg-firmware #49: HTTP 302 forward to 192.168.1.1 if DNS host is used to call web UI

stevenengland, 2024-11-13:
Hey folks, unfortunately the Telegram Chat always deletes my messages so I ask my question here:

When I call the web UI via IP all is good. If I call it via DNS it is redirecting me to 192.168.1.1

Do I miss something in the config or is it a bug?

[image]

Version used: 20240519

stevenengland, 2024-11-16:
Hi @xyzroe, the DNS name is a locally resolved one. What do you mean by 0914? I do not see any code fragment that would make a difference when I change the DNS entry to point from 0914.xyz.tld to this host.

KR

xyzroe, 2024-11-16:
Locally resolved? You use custom DNS ? So you should check your DNS settings. It is not related to XZG. 

0914 - version name of firmware

xyzroe, 2024-11-16:
I think you're using some proxy, not just a DNS name.

stevenengland, 2024-11-16:
Hi @xyzroe, please have a look at the image I provided. The DNS name is resolved to the actual IP in my network (ending .72). And the webserver is answering with a 302 forward to 192.168.1.1 None of this is related to my network. And no, there is no proxy in front of UZG.

xyzroe, 2024-11-16:
[my_setup.jpg](https://github.com/user-attachments/assets/55a70a0a-dbad-412c-b623-cb7d3e6719e0)

xyzroe, 2024-11-16:
[image]

 🤔 and how does it work on my network?

stevenengland, 2024-11-16:
Maybe because there is a config option interferring. Wouldn't you agree that 192.168.1.1 is a magic IP that fits perfectly fine to the ones that UZG uses in different defaults? Why ever would something in my network where I do not even operate this subnet forward to this IP?

stevenengland, 2024-11-16:
And again: IP .72 is the UZG. If I call it by this IP directly everythings fine

stevenengland, 2024-11-16:
What I will try though is to update (didn't realize the newer version even though you mentioned it, sorry).

xyzroe, 2024-11-17:
> 192.168.1.1 is a magic IP that fits perfectly fine to the ones that UZG uses in different default 

Of course no. Cause these defaults are only in web, not in config of UZG. And even in AP mode UZG uses *.*.4.1 not *.*.1.1

xyzroe, 2024-11-20:
I also use pihole. And you can see the screen.

stablestud, 2024-12-14:
Hi @xyzroe, I'm having the same issue after updating to 1.0.0

Usually I access the coordinator via the domain zigbee.internal, but now its being redirected to 192.168.1.1. 
This seems to be a bug in the newest version

See the below curl output:
```
~$ curl zigbee.internal -v -L
* Host zigbee.internal:80 was resolved.
* IPv6: (none)
* IPv4: 192.168.144.12
*   Trying 192.168.144.12:80...
* Connected to zigbee.internal (192.168.144.12) port 80
> GET / HTTP/1.1
> Host: zigbee.internal
> User-Agent: curl/8.9.1
> Accept: */*
> 
* Request completely sent off
< HTTP/1.1 302 Found
< Content-Type: text/plain
< Location: http://192.168.1.1
< Content-Length: 0
< Connection: close
< 
* shutting down connection #0
* Issue another request to this URL: 'http://192.168.1.1/'
*   Trying 192.168.1.1:80...
```

The devices redirects for no good reason to 192.168.1.1

xyzroe, 2024-12-14:
install XZG firmware and try again. 

there isn't version 1.0.0 in XZG, so I think you're using ZigStar firmware, which is not developing more.

stablestud, 2024-12-15:
Yes this fixed the issue.
After v1.0.0 I flashed v20240914 and now it works again =)

closed 2024-12-15 by xyzroe